ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #23: CORREGIR Declaracion de c[] de manera dinamica para que se desplieguen bien los string Largos

ChalsLapWork, 2025-02-26:


ChalsLapWork, 2025-02-26:
debug  26-feb-25 341pm disp_RevE2.21 RUNNING. Testing String Largos
modificamos el case3 agregamos esta linea 
                                datos_len-=3;//para el procesamiento    
antes que otra cosa

ChalsLapWork, 2025-02-26:
debug 26-feb-25 357pm disp_RevE2.22 RUNNING. Testing String Largos
modificamos el case3 agregamos esta linea
datos_len-=3;//para el procesamiento
antes que otra cosa, error de sintaxis

ChalsLapWork, 2025-02-26:
debug 26-feb-25 357pm disp_RevE2.22 RUNNING. Testing String Largos
modificamos el case3 agregamos esta linea
datos_len-=3;//para el procesamiento
antes que otra cosa, error de sintaxis

Todavia Aparece el string
en la linea 10 del menu Portal Inicio:
S             PL
                 ^----No debe Aparecer
Verificar y Probar esta Linea:                                 datos_len-=3;//para el procesamiento    
aumentando o disminuyendo el numero en la funcion:
   VFDserial_SendBlock_Tx   en case 3

chalsRemoteServer, 2025-02-27:
debug 26-feb-25 945pm disp_RevE2.24 RUNNING. 
probando porque se mandan los comandos dos veces y probando los strings largos

chalsRemoteServer, 2025-02-27:
El problema esta en el semaforo sem vacios porque se incrementa hasta 64 siempre en la funcion productor y el fifo circular llena y llena la fifo indefinidamente  hay un mal control con los semaforos lleno y vacio cuando se llena la buffer_circular

ChalsLapWork, 2025-03-10:
ya quedaron las funciones de procesar comando y consumidor en lo que respecta que ya reconoce una cadena de comando

closed 2025-03-10 by ChalsLapWork