ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #24: Semaforos Llenos y Vacios  estan mal controlados se sobre-escribe la buffer_Circular

chalsRemoteServer, 2025-02-27:
los buffers llenado y vacios se cuentan y descuentam mal en la funcion productor y sendBlock_buf, por eso se sobreescribe buffer_circular cuando se llena en 10 y se sobreescribe  en la posicion 1 en lugar de esperar el hilo y detenerse ha que haya espacio.

chalsRemoteServer, 2025-02-27:
debug 26-feb-25 1037pm revisando, Must-Stop-Running se necesita cambiar codigo para corregir el issue

ChalsLapWork, 2025-02-27:
debug 27-feb-25 0125pm disp_RevE2.25 RUNNING, status: Display Portal Inicio: no Completo, 
los semaforons de decrementan e incrementan de manera erronea y hace que la fifo circular se sobreescriba, cambiamos la funcion productor y movimos los semaforos despues del memcopy, vamos a probar

ChalsLapWork, 2025-02-27:
debug 27-feb-25 230pm disp_RevE2.26 RUNNING, status: Display Portal Inicio: no Completo,
los semaforons de decrementan e incrementan de manera erronea y hace que la fifo circular se sobreescriba, cambiamos la funcion productor y movimos los semaforos despues del memcopy, vamos a probar, 
Volvemos a mover los semaforos, no se ha corregido el error.

ChalsLapWork, 2025-02-27:
debug 27-feb-25 546pm disp_RevE2.27 RUNNING, status: Display Portal Inicio: no Completo,
reestruturamos y recodificamos el hilo productor con estados y switches, vamos a probar si ya no se reesccriben la fifo circular y los semaforos funcionan bien

chalsRemoteServer, 2025-02-28:
debug 27-feb-25 744pm disp_RevE2.28 RUNNING, status: Display Portal Inicio: no Completo, 
seguimos modificando el productor tenia el problema que no aumentaba el buffer2-len vamos a probar este nuevo codigo

chalsRemoteServer, 2025-02-28:
debug 27-feb-25  949pm disp_RevE2.29 RUNNING, status: Display Portal Inicio: ,
corrio  y funcionaron los semaforos y los hilos prod cons

chalsRemoteServer, 2025-02-28:
debug 27-feb-25 100pm disp_RevE2.30 RUNNING, status: Display Portal Inicio corriendo bien, agregamos cuadromadre vamos aprobarlo

ChalsLapWork, 2025-02-28:
debug 27-feb-25 548pm disp_RevE2.27 RUNNING, status: Display Portal Inicio: no Completo,
reestruturamos y recodificamos el hilo productor con estados y switches, vamos a probar si ya no se reesccriben la fifo circular y los semaforos funcionan bien, error de sintaxis corregido

ChalsLapWork, 2025-02-28:
issue resuelto 28-feb-25 252pm:
Los fifos e hilos de VFD estan corriendo y funcionando correcto,  disp_RevE2.30, branch: RevE2_VFD_func,.

closed 2025-02-28 by ChalsLapWork