ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #30: Crear Hilo de Comunicaciones a Tarjeta Procesadora

ChalsLapWork, 2025-03-03:
Se necesita Crear el Hilo de comunicaciones de la tarjeta procesadora a la Rpi., 

se va recibir una serie de strings los cuales van a ser extraidos y reconocidos como comandos y estos comandos van a ser metidos a una fifo circlular. que va ser procesada por el hilo protocolo.

chalsRemoteServer, 2025-03-04:
debug 3-mar-25  1109pm., branch:Feature RevE2 commProc  serial.01,
iniciamos despliegue del primer ejecutable de pruebas del serial receptor.

chalsRemoteServer, 2025-03-04:
debug 3-mar-25 1136pm., branch:Feature RevE2 commProc serial.01,
error de libreria no existe wiringpiserial

chalsRemoteServer, 2025-03-04:
debug 3-mar-25 1218am, branch:Feature RevE2 commProc serial.03
eprobando otravez corriendo programa

ChalsLapWork, 2025-03-10:
10-mar.25 5:34pm ,  branch Featura RevE2 commProc, ya se reconoce una cadena de comando y se despliega un string que dice comando aceptado, con basura antes y despues, y sono una vez sin falla o muchas veces,

closed 2025-03-10 by ChalsLapWork