ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #31: Error en la comunicacion, no se REcibe lo que se Transmite

ChalsLapWork, 2025-03-05:
debug 5-mar.25 120pm: branch: Feature RevE2, 
La recepcion no se recibe bien con lo que se transmite usando una consola de descklight,, hicimos varios scripts para probar el puerto serial pero no funciona bien la recepcion,  hicimos un programa con python y este si funciona bien la TX y RX con un puente, se recibe lo que se transmite, ahora vamos a pasar el codigo del programa en python que si funciono bien para probarlo con libreria wiringPi en C, 

los scripts en ssh bash tienen un error que no cierrarn el puerto ni matan el proceso aunque te salgas del programa con el prompt

ChalsLapWork, 2025-03-05:
debug 5-mar.25 1:30pm: branch: Feature RevE2,
se sigue modificando el programa en C para recibir y enviar serial, vamos a probar otravez porque en C no funciona, en python si funciona

chalsRemoteServer, 2025-03-06:
The Serial corrupt bytes is due to the core clock set as dynamic, must change to  fix value to 250.
 sudo nano /boot/config.txt
add lines
dtoverlay=disable-bt
dtoverlay=miniatura-bt
core_freq=250
enable_uart=1
force_turbo=1
Para ver la frecuencia del núcleo, puedes usar el siguiente comando:
vcgencmd measure_clock arm
con este comando:
ls -l /dev/serial0
debe dar 
[permisos] 1 root root  fecha  /dev/serial0 -> ttyAMA0 
si no apunta a ttyAMA0  esta como dinamico el reloj y debe  de dirigirse a ttyAMA0

closed 2025-03-06 by chalsRemoteServer

ChalsLapWork, 2025-03-06:
debug 5-mar.25 1:42pm: branch: Feature RevE2,
el programa de pruebas en C sigue sin funcionar, vamos a probar unos cambios

ChalsLapWork, 2025-03-06:
¿Cómo fijarlo si aún no está en 250 MHz?

Si el reloj sigue cambiando, revisa el archivo de configuración con:

cat /boot/config.txt | grep core_freq

Debe contener:

core_freq=250

Si no está, agrégalo manualmente:

sudo nano /boot/config.txt

Agrega o edita esta línea:

core_freq=250

Guarda los cambios (Ctrl+X, luego Y, luego Enter) y reinicia la Raspberry Pi:

sudo reboot

Luego, vuelve a ejecutar vcgencmd measure_clock core para confirmar que ahora está fijo en 250 MHz.