ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #36: Error de ciclado de DAtos de Buffer de Recepcion

chalsRemoteServer, 2025-03-10:
la funcion cosumidor serial processor  no mete el buffer6 size a la  funcion proceamiento de cadena serproc y el while de procesamiento de cadena serproc ese while  nunca ve el cero, asi que tenemos que poner el tamaño del buffer y tambien el ultimo en la func consumidor, poner despues de strncpy un cero asi que

1.- poner en el buffer6 despues de strncpy en el ultimo que se copio a cero
2.- meter ala funcion  proc cadena ser proc el  tamaño del buffer
3 dentro de la func. proc. de cadena serProc. el while para que tambien evalue el tamaño de la cadena si nunca ve el '/0'

ChalsLapWork, 2025-03-10:
10-,mar-25 942am Branch Feature-RevE2-CommProc status: Running,
se agrego esta linea en la funcion:  while((*(c+indice)!='\0')||(indice<BUFFER6_SIZE)){
 func:Procesamiento_de_cadena_serProc porque se sigue ciclando cuando se detecta un comando valido.

ChalsLapWork, 2025-03-10:
10-,mar-25 1151am Branch Feature-RevE2-CommProc status: Running,
errorres de sintaxis, seguimos probando la linea del commit anterior

ChalsLapWork, 2025-03-10:
10-,mar-25 0114am Branch Feature-RevE2-CommProc status: Running.
Se tiene  que la variable indice es uchar y el  size_buffer6 es usint, y por eso nunca alcanza el final., se necesita cambiar el uchar hacia usint codigo cambiado:
  static unsigned short int indice; //el buffer6 es  de 1024 tamaño

ChalsLapWork, 2025-03-10:
10-,mar-25 01:42am Branch Feature-RevE2-CommProc status: Running.
vamos a poner un indicador en la funcion Procesamiento_de_cadena_serProc para ver el avance de procesamiento de la cadena para hacer mas facil la depuracion;

ChalsLapWork, 2025-03-10:
10-,mar-25 01:45am Branch Feature-RevE2-CommProc status: Running.
vamos a poner un indicador en la funcion Procesamiento_de_cadena_serProc para ver el avance de procesamiento de la cadena para hacer mas facil la depuracion;
error de sintaxis

ChalsLapWork, 2025-03-10:
10-,mar-25 01:55am Branch Feature-RevE2-CommProc status: Running.
hay error de acoplamiento de codigo gdb con gcc vamos a recompilar todo

ChalsLapWork, 2025-03-10:
10-,mar-25 0405am Branch Feature-RevE2-CommProc status: Running.
Has sido reconocido el protocolo con el comando 03 06 11 01 02 03 04 EA 02 pero si ponemos cadenas una al lado de otra puede occurrir una desincronia por el 03 y 02 que hay como dato, asi que se debe tener que poner al principio de la cadena unos bytes de sincronia como e.g. 0xFF hasta que tengamos un protocolo mas robusto., tambien se va a necesitar que el que manda un comando el receptor conteste de ACK del comando y cierre el asunto sino que lo vuelva a enviar sino se va perder datos.

closed 2025-03-10 by ChalsLapWork