ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #38: Procesar Comando Serial uProc. de despleigue de Barra de Deteccion en Portal Inicio

ChalsLapWork, 2025-03-10:
el comando serial que se envia de la tarjeta procesadora que sea procesado captado y que ese valor de la barra de deteccion que se eleve la barrita de deteccion a ese valor que la procesadora comanda.

chalsRemoteServer, 2025-03-11:
debug: 10-mar-25 1052pm Branch: Feature_RevE2_CommProc, status: Merging 
Se va juntar/mezclar el codigo del ejemplo06 serial funcionando con reconocimiento de comando,

chalsRemoteServer, 2025-03-11:
debug: 10-mar-25 1052pm Branch: Feature_RevE2_CommProc, status: Merging
Se va juntar/mezclar el codigo del ejemplo06 serial funcionando con reconocimiento de comando,,
error de sintaxis.

chalsRemoteServer, 2025-03-11:
debug: 10-mar-25 1135pm Branch: Feature_RevE2_CommProc, status: Building
se hizo un makefile vamos a probarlo.

chalsRemoteServer, 2025-03-11:
debug: 10-mar-25 1135pm Branch: Feature_RevE2_CommProc, status: Building
se corrigieron algunos errores  de sintaxis

chalsRemoteServer, 2025-03-11:
debug: 10-mar-25 1135pm Branch: Feature_RevE2_CommProc, status: Building
se agregaron  las librerias en el archivo serial.c

chalsRemoteServer, 2025-03-11:
debug: 11-mar-25 1213am Branch: Feature_RevE2_CommProc, status: Building
falta declarar los hilos en serial.h, pero no se ha hecho.

ChalsLapWork, 2025-03-11:
debug: 11-mar-25 0850am Branch: Feature_RevE2_CommProc, status: Debugging new code, 
se agrego declaracion de func. de serial que faltaban

ChalsLapWork, 2025-03-11:
debug: 11-mar-25 0850am Branch: Feature_RevE2_CommProc, status: Debugging new code,
No se esta ejecutando el menu y el despliegue del menu, se queda ciclado en el init del serial, vamos a poner init display antes de serial. y a poner las declaracion de limpieza de puerto e hilos en main en cleanup

ChalsLapWork, 2025-03-11:
debug: 11-mar-25 0924am Branch: Feature_RevE2_CommProc, status: Debugging new code,
No se esta ejecutando el menu y el despliegue del menu, se queda ciclado en el init del serial, vamos a poner init display antes de serial. y a poner las declaracion de limpieza de puerto e hilos en main en cleanup.. errores de sintaxis

ChalsLapWork, 2025-03-11:
debug: 11-mar-25 0337pm Branch: Feature_RevE2_CommProc, status: Debugging new code,
se agrego el codigo en transmisor TX de VFD para que despleigue las cajas, se modifico codigo en varios archivos

ChalsLapWork, 2025-03-11:
debug: 11-mar-25 0337pm Branch: Feature_RevE2_CommProc, status: Debugging new code,
se recalculo  varios funciones, para el despliegue VFD,

chalsRemoteServer, 2025-03-15:
debug: 14-mar-25 0842pm Branch: Feature_RevE2_CommProc, status: Debugging new code,
Regresamos a programar despues de caer en las garras del COVID,  no quedamos en que...
hicimos codigo  en Tx y procesos anteriores para  que se pinten las cajas de la barra de deteccion, 
vamos aprobar el codigo para pintar las cajas.

chalsRemoteServer, 2025-03-15:
debug: 14-mar-25 0842pm Branch: Feature_RevE2_CommProc, status: Debugging new code,
reacomodamos el codigo para seguir depurando el comando 11h que es el barra comando

chalsRemoteServer, 2025-03-15:
debug: 14-mar-25 0950am Branch: Feature_RevE2_CommProc, status: Debugging new code,
Modificamos  la funcion Transmisor VFD en el case CMD_BAR porque aun no se transmite bien el  comando de las cajas y no se pintan sino que hay caracteres en lugar de cajas,

chalsRemoteServer, 2025-03-15:
debug: 15-mar-25 1015am ver.40.0.2.1 Branch: Feature_RevE2_CommProc, status: Debugging new code,
getBoxPattern <--funcion modificada

chalsRemoteServer, 2025-03-15:
debug: 15-mar-25 1150am ver.40.0.2.3 Branch: Feature_RevE2_CommProc, status:RUNNING
se modifico de 100mil a 25mil ya que sigue lento como incrementan las cajas en la barra de deteccion.
const unsigned int TIEMPO_CAJAS=25000;//useg tiempo de espera para cambio de cajas

chalsRemoteServer, 2025-03-15:
debug: 15-mar-25 1155am ver.40.0.2.4 Branch: Feature_RevE2_CommProc, status:RUNNING
se modifico de 25mil a 12mil, sigue lento,  ya que sigue lento como incrementan las cajas en la barra de deteccion.
const unsigned int TIEMPO_CAJAS=12000;//useg tiempo de espera para cambio de caja

chalsRemoteServer, 2025-03-17:
debug 16-mar-25 1130pm  ver.40.0.3.1  Branch Feature_RevE2_CommProc, status: Running,
fallo la comunicacion serial cuando hicimos el merge, ya no funcinoa mas el serial y tenemos que recosntruirlo desde cero porque no se administraron bien las bransch no tenemos un ejemplo funcionando perfecto.

chalsRemoteServer, 2025-03-17:
debug 16-mar-25 1205pm ver.40.0.3.2  Branch Feature_RevE2_CommProc, status: Running,
modificamos el codigo del nuevo ejemplo vamos a correr el codigo,

chalsRemoteServer, 2025-03-18:
debug debug 17-mar-25 0827pm ver.40.0.3.2 Branch Feature_RevE2_CommProc, status: Running,
Se termino el problema, el cual es que los ejecutables no se meuren cuando volvemos a compilar el ejemplo despeus de modificarlo y el puerto serial se bloquea. el puerto serial y el ejemplo esta funcionando bien tenemos el comando de bara de deteccion funcionando por comandos seriales

closed 2025-03-18 by chalsRemoteServer

chalsRemoteServer, 2025-03-18:
17-mar-25 843pm: debug ver.40.0.3.3 Branch Feature_RevE2_CommProc, status: Running,
esta funcion ya esta trabajando correctamente,