ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #39: Hacer pruebas de Reconocimiento de recepcion de comando de despliegue de posicion de barra de deteccion

ChalsLapWork, 2025-03-10:
que el comando de recepcion serial que proviene de la procesadora sea reconocido con y sin basura antes y despues, en modo masivo y en modo intermiente escaso si falla y sin perdida de veces

chalsRemoteServer, 2025-03-16:
16-mar-25 1256pm ver.40.0.2.14 Branch feature revE2_commProc. hemos hecho la mezca/merge de rama E4,  haora vamos a hacer la ultima prueba de mover la barra de deteccion con comandos seriales.

chalsRemoteServer, 2025-03-16:
16-mar-25 1256pm ver.40.0.2.14 Branch feature revE2_commProc. hemos hecho la mezca/merge de rama E4, haora vamos a hacer la ultima prueba de mover la barra de deteccion con comandos seriales. Error de sintaxis..

chalsRemoteServer, 2025-03-16:
16-mar-25 1256pm ver.40.0.2.145 Branch feature revE2_commProc. 

Estamos haciendo ultimas pruebas pero dejo de funcionar la recepcion serial necesitamos poner mas printf para evaluar.

chalsRemoteServer, 2025-03-18:
Debug 17-Mar-25 0836pm: se hicieron pruebas de reconocimiento de comandos de despliegu de barra de deteccion comandos:
con STX al inicio y ETX el final,  los valores de la barra son 0,20,50,80,99% los valores de los comandos correspondientes fueron:
 03 11 00 d4
03 11 14 b8
03 11 32 4a
03 11 50 63
03 11 63 fa

closed 2025-03-18 by chalsRemoteServer