ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #44: Crear La estructura de datos de Producto,

chalsRemoteServer, 2025-03-18:
crear la estructura de datos de Producto para poner la variable global conteo_de_rechazos que necesitamos para desplegarlo en portal inicio

ChalsLapWork, 2025-03-18:
18-mar.-25 951am: ver:40.0.3.5 Branch:Feature_RevE5_Cuent_Rech, status:Building code.
se construye codigo para poner la estructura de datos de Producto, se va a poner en el archivo queue donde estan todas las estructuras de datos,
se agrego :
   unsigned short int Cuenta_Rechazos;//numer de rechazos de producto
 que es la variable que va contar el numero de rechasos que ha tenido el producto, contados apartir de la barra de deteccion,
y se agrego el numero de cuenta de produtos.

La estructura quedo asi:
struct _PRODUCTO_{
   unsigned short int Cuenta_Rechazos;//numer de rechazos de producto
   unsigned short int Cuenta_Productos;//numero de productos que ha pasado por detector
   unsigned short int Sensibilidad;//configurada para este producto
   unsigned char phase;//phase del producto, numero entero
   float fase;//fase del producto numero flotante
   unsigned char Phasefrac;// faccion del numero de fase.
};

ChalsLapWork, 2025-03-18:
18-mar.-25 1006am: ver:40.0.3.5 Branch:Feature_RevE5_Cuent_Rech, status:Building code.
error de sintasis menor.

ChalsLapWork, 2025-03-18:
18-mar.-25 1015am: ver:40.0.3.5 Branch:Feature_RevE5_Cuent_Rech, status:Building code.
el nombre de producto en sus variantes ya estaba ganado

closed 2025-03-18 by ChalsLapWork