ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #54: Crear codigo para que evalue la barra de deteccion y calcule num de rechazos

ChalsLapWork, 2025-03-18:
que haya un control cuando la barra de deteccion supere el 50% se incremente el num de rechazos y este a su vez se despliegue en el portal inicio.

ChalsLapWork, 2025-03-18:
18-mar.-25 0358am: ver:40.0.3.9 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING
se creo las funciones necessariaas para evaluar los conteos de producto y rechazo en base a los comandos que llegan de barra de deteccion.

ya se creo la funcion:
void display_CuentaRechazosProducto(unsigned char deteccion){
unsigned char sel=0;	
const unsigned char POS_X_CONT_RECHAZ=100;
const unsigned char POS_X_CONT_PROD=100;
const unsigned char POS_Y_CONT_RECHAZ=POSY8;
const unsigned char POS_Y_CONT_PROD=POSY10;

if(deteccion>50){sel=1;
      producto2.Cuenta_Rechazos++;
	  producto2.Cuenta_Productos++;}
else{if(deteccion>10){sel=2;producto2.Cuenta_Productos++;}}
if(vfd.menu.contexto.Actual==PORTAL_INICIO){
	   switch(sel){
		  case 1:VFDserial_Sendusint(producto2.Cuenta_Rechazos,POS_X_CONT_RECHAZ,POS_Y_CONT_RECHAZ,CENTRAL);
		 		 VFDserial_Sendusint(producto2.Cuenta_Productos,POS_X_CONT_PROD,POS_Y_CONT_PROD,CENTRAL);
		         break;
		  case 2:VFDserial_Sendusint(producto2.Cuenta_Productos,POS_X_CONT_PROD,POS_Y_CONT_PROD,CENTRAL);
		        break;
	      default:break;}}			  
}//fin de desplaiegue de conteo de rechazos y producto++++++++++++++++++++++++++

ChalsLapWork, 2025-03-18:
18-mar.-25 0403am: ver:40.0.3.10 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING,probando en portal inicio, cont prod & cont rechazo.

ChalsLapWork, 2025-03-18:
18-mar.-25 0431am: ver:40.0.3.11 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING,probando en portal inicio, cont prod & cont rechazo.
Depurando de errore menores pero dificiles de detectar.

ChalsLapWork, 2025-03-19:
18-mar.-25 0601am: ver:40.0.3.11 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING,probando en portal inicio, cont prod & cont rechazo.
esatamos construyendo codigo en el TX vfd, modificando el case CMD_USINT

chalsRemoteServer, 2025-03-19:
19-mar.-25 1259am: ver:40.0.3.13 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING,probando en portal inicio, cont prod & cont rechazo. seguimos depurando el case CMD_USINT de TX vfd

chalsRemoteServer, 2025-03-19:
19-mar.-25 0745am: ver:40.0.3.15 Branch:Feature_RevE5_Cuent_Rech, status:DEBUGGING,
hay un error en serial.c: switch valores de la var se meten a la funcion rechazos producto cntador  corregir el switch.

ChalsLapWork, 2025-03-19:
19-mar.-25 0916am: ver:40.0.3.16 Branch:Feature_RevE5_Cuent_Rech, status:RUNNING
este issue ha sido completado, ya calcula rechazos y barra deteccion funciona ok, pero hay algunos cosas que corregir las cuales ponemos en otros issues

closed 2025-03-19 by ChalsLapWork