ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #70: Crear los SubMenus de MEnu Insight

ChalsLapWork, 2025-03-28:
Crear los submenus de Menu insight
par apoder poner los comandos de LEFT, RIGHT

chalsRemoteServer, 2025-03-30:
28-mar.-25 512pm: ver:40.0.3.59 Branch:Feature_RevE7 status: Debug,
Construyendo los  submenus del menu INSIGHT., en proceso de construccion.

ChalsLapWork, 2025-03-31:
31-mar.-25 952am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode

ChalsLapWork, 2025-03-31:
31-mar.-25 952am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de los submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 1120am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de los submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 1121am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de los submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 1214am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de los submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 1244am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 331am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 414pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-03-31:
31-mar.-25 530pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
31-mar.-25 600pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 959am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 1018am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 1110am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 1125am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 325am: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 336pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 338pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 459pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 507pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 509pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 512pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-01:
1-abr.-25 552pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-02:
2-abr.-25 419pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-02:
2-abr.-25 432pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-03:
3-abr.-25 1102pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-03:
3-abr.-25 1109pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-03:
3-abr.-25 1129pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight

ChalsLapWork, 2025-04-03:
3-abr.-25 104pm: ver:40.0.3.60 Branch:Feature_RevE8 MenuInsight status: BuildCode
pasando el codigo de os submenus del menu insight