ChalsLapWork/Disp_IN2UP_RPi3A_RevB_gitHub #75: Agregar Contraseña a Ajuste de Producto

ChalsLapWork, 2025-04-03:
Agregar el menu de escribir contraseña para ingresar a Ajuste de Producto.

ChalsLapWork, 2025-04-03:
3-abr.-25 0544pm: ver:40.0.3.66 Branch:Feature_RevE8 MenuInsight status…
se esta modificando el codigo para que se abra el password menu para entrar a ajuste de prod.

ChalsLapWork, 2025-04-07:
4-abr.-25 0522pm: ver:40.0.3.67 Branch:Feature_RevE8 MenuInsight status…
vamos a probar el menu de poner contraseña para entrar al producto

ChalsLapWork, 2025-04-07:
7-abr.-25 0328pm: ver:40.0.3.68 Branch:Feature_RevE8 MenuInsight status…
vamos a probar el menu de poner contraseña para entrar al producto

ChalsLapWork, 2025-04-07:
7-abr.-25 0353pm: ver:40.0.3.68 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-07:
7-abr.-25 0353pm: ver:40.0.3.69 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-07:
7-abr.-25 0353pm: ver:40.0.3.70 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-07:
7-abr.-25 0545pm: ver:40.0.3.71 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-08:
7-abr.-25 0600pm: ver:40.0.3.71 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

chalsRemoteServer, 2025-04-08:
7-abr.-25 1050pm: ver:40.0.3.72 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-08:
8-abr.-25 0259am: ver:40.0.3.74 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-08:
8-abr.-25 0302am: ver:40.0.3.75 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-08:
8-abr.-25 0321am: ver:40.0.3.76 Branch:Feature_RevE8 MenuInsight status: Integrando Codigo

ChalsLapWork, 2025-04-09:
9-abr.-25 0532am: ver:40.0.3.77 Branch:Feature_RevE9  RevE9_Password  status: Integrando Codigo

ChalsLapWork, 2025-04-10:
9-abr.-25 0536am: ver:40.0.3.77 Branch:Feature_RevE9 RevE9_Password status: Integrando Codigo.
Tenemos funcionando acceso al menu de Clave de acceso desde el MenuIInsight->ClavedeAcceso->ConfigProd

ChalsLapWork, 2025-04-15:
15-abr.-25 3:04pm: ver:40.0.3.98 Branch:Feature_RevE9 RevE9_db_Password.…
se ha hechos mucho codigo para sacar las claves de acceso del archivo sistema.ini  con todos sus controles, tener funcionando el control de eventos con el archivo log.txt, y ya funciona pasar al menu setup-product con password de por en medio.

closed 2025-04-15 by ChalsLapWork